victorlucachi/void9 #1: Column pinout error and improvement

fleaz, 2021-05-24:
Hey Victor,

first of all big thanks for the Void9! Already build one myself and gathered the parts for two sets as presents for some friends of mine :)

But I found a small error in your handwiring guide on your website and related to this I would suggest an improvement:
On your website the pinout for the columns are swapped (It must be 1,0,4 not 0,1,4). In relation to this I would suggest to use the pin names which are printed on the board itself instead of using the 'internal names' which would make assembly way more user friendly in my opinion. Because while building the void9 I first had to google a pinout graphics of the Pro Micro which uses the internal name so I know which pads on the board to use.

With the printed names the mapping would become this:

|     | Row0 | Row1 | Row2 | Col0 | Col1 | Col2 |
|-----|------|------|------|------|------|------|
| Pin | A3   | A2   | A1   | 2    | 3    | 4    |

Best,
Felix

victorlucachi, 2021-07-31:
Hello Fleaz, my apologies for the very, very late reply.

Glad to hear that that you've built a bunch of them, thank you! I appreciate you bringing the column error to my attention, I will fix it as soon as I have the chance, hopefully by the end of next week.

I will also make use of your suggestion and at the very least I will include the alternate pin names, maybe even a pinout graph.

Cheers!

impoze, 2021-09-29:
first of all, thanks for providing the files & very clear guide.

I originally had some trouble with the board not working after flashing and wiring everything up and have spotted the issue.

In the wiring guide, the pin out is incorrect in step 8.

Column0 is marked as D0
Column1 is marked as D1

[image]

On the github it is documented correctly

[image]

shanselman, 2023-01-24:
> Hey Victor,
>
> first of all big thanks for the Void9! Already build one myself and gathered the parts for two sets as presents for some friends of mine :)
>
> But I found a small error in your handwiring [...]

Thanks! I got stuck here as well, thanks for this guide...this matches my printed pinouts. I didn't know the internal names.

closed 2023-06-13 by victorlucachi