victorlucachi/void9 #3: Type-c compatible

Bischi1997, 2022-02-11:
Hi @victorlucachi 

Can you maybe add the source files (eg. Fusion 360), so I could create an updated version for an Arduino micro with type-c (eg. Tenstar Robot pro micro)?
I'd be happy to add it via pull request, once done :)

Regards
Bischi

victorlucachi, 2022-02-11:
Hey,

Thanks for reaching out. I'll be releasing updated versions for my designs, which will include modular controller support among other things, in the near future; unfortunately they wont be open source.

Care to go into detail regarding the angled version, defective in what way? I can download and slice the file without any issues.

Bischi1997, 2022-02-11:
Cool to hear that an update is coming. Any rough estimate when the eta will be?

Just tried it again, seems to work now, no Idea what I did wrong last time :)

victorlucachi, 2022-02-11:
Hopefully in a week or two. I've been working on the files for a while now and they're mostly ready, I just have to re-print all of them to make sure that I didn't overlook anything.

Sweer, glad it worked.

Bischi1997, 2022-02-11:
Sure, let me know if I can test-print it for you as well.
Still waiting for my keyswitches, other than that I have all the components ready.

Bischi1997, 2022-04-02:
Hi Victor
Any Update?

victorlucachi, 2022-04-17:
Sorry for the late(ish) reply, I've been away for the past few weeks. Soon🤞

[image]

Sparticuz, 2023-05-25:
Any progress on this? I've got a waveshare rp2040 that I'd like to use in it

Bischi1997, 2023-06-13:
Seems like the answer is soon™

victorlucachi, 2023-06-13:
_soonish_.

I will update the repos either today or tomorrow, the files have been printed and tested before, but keep in mind that there might be issues.

I will have to rework the controller mount before posting the source files because in its current state, even though it's a parametric design, it sometimes breaks and the resulting models are unusable.

having said that, is this the controller that you plan on using @Sparticuz?

[image]

Sparticuz, 2023-06-13:
Yep, that's it. I've also got the qmk and via files to build the firmware for it and have been planning on posting them.

victorlucachi, 2023-06-13:
> Yep, that's it. I've also got the qmk and via files to build the firmware for it and have been planning on posting them.

Awesome. Reworked macropad files for the top/bottom are already up, I will add the controller mount shortly.

closed 2023-06-13 by victorlucachi