victorlucachi/void9 #4: QMK compilation error

cgrs, 2022-04-07:
When using your QMK fork in order to support this macropad, QMK does not recognize the folder of the keyboard firmware with the following error:
```bash
$ # using make
$ make handwired/void9:default
QMK Firmware 0.11.69
usage: qmk list-layouts [-h] [-kb KEYBOARD]
qmk list-layouts: error: argument -kb/--keyboard: invalid keyboard_folder value: 'handwired/void9'
$ # using qmk compile
$ qmk compile -kb handwired/void9 -km default
usage: qmk compile [-h] [-c] [-e ENV] [-j PARALLEL] [-n] [-km KEYMAP] [-kb KEYBOARD] [filename]
qmk compile: error: argument -kb/--keyboard: invalid keyboard_folder value: 'handwired/void9'
```

I've found the issue is in the `DEFAULT_FOLDER` directive in the `rules.mk` file, which somehow messes with the QMK compilation process:
```diff
--- a/keyboards/handwired/void9/rules.mk
+++ b/keyboards/handwired/void9/rules.mk
@@ -25,4 +25,4 @@ NKRO_ENABLE = no            # USB Nkey Rollover
 MIDI_ENABLE = no            # MIDI support
 FAUXCLICKY_ENABLE = no      # Use buzzer to emulate clicky switches

-DEFAULT_FOLDER = handwired/void9/default
\ No newline at end of file
+# DEFAULT_FOLDER = handwired/void9/default
```
Once removed, the firmware compiles just as fine (with some warnings).

victorlucachi, 2022-04-17:
sorry for the late reply, and thank you for the fix!

I've been meaning to go over all of my builds for a long while now, clean up the files and fix any issues, but never got around to doing it; hopefully in the near future.

r4ph74, 2022-05-15:
I already removed the default folder in rules. mk 
I still have a problem when compiling  as follow 

[r4ph@DESKTOP-H1TUS21 qmk_firmware]$ qmk compile -kb handwired/void9 -km default
Ψ Compiling keymap with make --jobs=1 handwired/void9:default

QMK Firmware 0.11.69
☒ handwired/void9: Number of keys for LAYOUT_ortho_3x3 does not match! info.json specifies 30 keys, C macro specifies 9
Making handwired/void9 with keymap default

☒ handwired/void9: Number of keys for LAYOUT_ortho_3x3 does not match! info.json specifies 30 keys, C macro specifies 9

 * BOOTMAGIC_ENABLE=lite                                                                               [ERRORS]

 *  is not a valid type of magic                                                                       [ERRORS]

builddefs/common_features.mk:558: *** Invalid BOOTMAGIC_ENABLE.  Stop.
Make finished with errors
make: *** [Makefile:413: handwired/void9:default] Error 1

victorlucachi, 2022-05-16:
> 

could you also change `BOOTMAGIC_ENABLE = lite` to `BOOTMAGIC_ENABLE = yes`, in rules.mk?

r4ph74, 2022-05-16:
I haven't . so I change it to BOOTMAGIC_ENABLE = yes in rules.mk 

and run compile again . 
------------------
Ψ Compiling keymap with make --jobs=1 handwired/void9:default

QMK Firmware 0.11.69
☒ handwired/void9: Number of keys for LAYOUT_ortho_3x3 does not match! info.json specifies 30 keys, C macro specifies 9
Making handwired/void9 with keymap default

☒ handwired/void9: Number of keys for LAYOUT_ortho_3x3 does not match! info.json specifies 30 keys, C macro specifies 9
avr-gcc.exe (GCC) 8.4.0
Copyright (C) 2018 Free Software Foundation, Inc.
This is free software; see the source for copying conditions.  There is NO
warranty; not even for MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.

Generating: .build/obj_handwired_void9/src/info_config.h
  [WARNINGS]
 |
 | ☒ handwired/void9: Number of keys for LAYOUT_ortho_3x3 does not match! info.json specifies 30 keys, C macro specifies 9
 |
Generating: .build/obj_handwired_void9/src/default_keyboard.h
  [WARNINGS]
 |
 | ☒ handwired/void9: Number of keys for LAYOUT_ortho_3x3 does not match! info.json specifies 30 keys, C macro specifies 9
 |
Generating: .build/obj_handwired_void9/src/layouts.h
  [WARNINGS]
 |
 | ☒ handwired/void9: Number of keys for LAYOUT_ortho_3x3 does not match! info.json specifies 30 keys, C macro specifies 9
 |
Compiling: keyboards/handwired/void9/void9.c
  [OK]
Compiling: keyboards/handwired/void9/keymaps/default/keymap.c
  [OK]
Compiling: quantum/quantum.c
  [OK]
Compiling: quantum/send_string.c
  [OK]
Compiling: quantum/bitwise.c
  [OK]
Compiling: quantum/led.c
  [OK]
Compiling: quantum/action.c
  [OK]
Compiling: quantum/action_layer.c
  [OK]
Compiling: quantum/action_tapping.c
  [OK]
Compiling: quantum/action_util.c
  [OK]
Compiling: quantum/eeconfig.c
  [OK]
Compiling: quantum/keyboard.c
  [OK]
Compiling: quantum/keymap_common.c
  [OK]
Compiling: quantum/keycode_config.c
  [OK]
Compiling: quantum/sync_timer.c
  [OK]
Compiling: quantum/logging/debug.c
  [OK]
Compiling: quantum/logging/sendchar.c
  [OK]
Compiling: quantum/bootmagic/bootmagic_lite.c
  [OK]
Compiling: quantum/bootmagic/magic.c
  [OK]
Compiling: quantum/matrix_common.c
  [OK]
Compiling: quantum/matrix.c
  [OK]
Compiling: quantum/debounce/sym_defer_g.c
  [OK]
Compiling: quantum/main.c
  [OK]
Compiling: quantum/process_keycode/process_magic.c
  [OK]
Compiling: quantum/process_keycode/process_grave_esc.c
  [OK]
Compiling: quantum/process_keycode/process_space_cadet.c
  [OK]
Assembling: platforms/avr/xprintf.S
  [OK]
Compiling: platforms/avr/printf.c
  [OK]
Compiling: tmk_core/protocol/host.c
  [OK]
Compiling: tmk_core/protocol/report.c
  [OK]
Compiling: tmk_core/protocol/usb_device_state.c
  [OK]
Compiling: tmk_core/protocol/usb_util.c
  [OK]
Compiling: platforms/suspend.c
  [OK]
Compiling: platforms/avr/platform.c
  [OK]
Compiling: platforms/avr/suspend.c
  [OK]
Compiling: platforms/avr/timer.c
  [OK]
Compiling: platforms/avr/bootloaders/caterina.c
  [OK]
Compiling: tmk_core/protocol/lufa/lufa.c
  [OK]
Compiling: tmk_core/protocol/usb_descriptor.c
  [OK]
Compiling: lib/lufa/LUFA/Drivers/USB/Class/Common/HIDParser.c
  [OK]
Compiling: lib/lufa/LUFA/Drivers/USB/Core/AVR8/Device_AVR8.c
  [OK]
Compiling: lib/lufa/LUFA/Drivers/USB/Core/AVR8/EndpointStream_AVR8.c
  [OK]
Compiling: lib/lufa/LUFA/Drivers/USB/Core/AVR8/Endpoint_AVR8.c
  [OK]
Compiling: lib/lufa/LUFA/Drivers/USB/Core/AVR8/Host_AVR8.c
  [OK]
Compiling: lib/lufa/LUFA/Drivers/USB/Core/AVR8/PipeStream_AVR8.c
  [OK]
Compiling: lib/lufa/LUFA/Drivers/USB/Core/AVR8/Pipe_AVR8.c
  [OK]
Compiling: lib/lufa/LUFA/Drivers/USB/Core/AVR8/USBController_AVR8.c
  [OK]
Compiling: lib/lufa/LUFA/Drivers/USB/Core/AVR8/USBInterrupt_AVR8.c
  [OK]
Compiling: lib/lufa/LUFA/Drivers/USB/Core/ConfigDescriptors.c
  [OK]
Compiling: lib/lufa/LUFA/Drivers/USB/Core/DeviceStandardReq.c
  [OK]
Compiling: lib/lufa/LUFA/Drivers/USB/Core/Events.c
  [OK]
Compiling: lib/lufa/LUFA/Drivers/USB/Core/HostStandardReq.c
  [OK]
Compiling: lib/lufa/LUFA/Drivers/USB/Core/USBTask.c
  [OK]
Compiling: tmk_core/protocol/lufa/usb_util.c
  [OK]
Linking: .build/handwired_void9_default.elf
  [OK]
Creating load file for flashing: .build/handwired_void9_default.hex
  [OK]
Copying handwired_void9_default.hex to qmk_firmware folder
  [OK]
Checking file size of handwired_void9_default.hex
  [OK]
 * The firmware size is fine - 13070/28672 (45%, 15602 bytes free)
--------------------

it ist done ?

r4ph74, 2022-05-16:
I check .build folder and I found this .

[image]

it should be ok?

victorlucachi, 2022-05-16:
> I check .build folder and I found this .
> 
> [image]
> 
> it should be ok?

yes. looks like it's all fine, flash the .hex with qmk toolbox

r4ph74, 2022-05-16:
Ok . Thank you very much for your help. 

I really appreciate it. 

it is compatible with Via right ? with uploading json file ?

victorlucachi, 2022-05-16:
> 

recompile the firmware with the `via_test` keymap if you want to use VIA Configurator

r4ph74, 2022-05-16:
Ok thanks .. I will try it. ..

r4ph74, 2022-05-16:
I have successfully flash the firmware. 
--------------------------------------
> avrdude.exe -p atmega32u4 -c avr109 -U flash:w:"H:\Scorpion\Project\3Dprinting\T3D\Mpad\QMK\qmk_firmware\.build\handwired_void9_via_test.hex":i -P COM18
> 
> Connecting to programmer: .
> Found programmer: Id = "CATERIN"; type = S
>     Software Version = 1.0; No Hardware Version given.
> Programmer supports auto addr increment.
> Programmer supports buffered memory access with buffersize=128 bytes.
> 
> Programmer supports the following devices:
>     Device code: 0x44
> 
> avrdude.exe: AVR device initialized and ready to accept instructions
> 
> Reading | ################################################## | 100% 0.00s
> 
> avrdude.exe: Device signature = 0x1e9587 (probably m32u4)
> avrdude.exe: NOTE: "flash" memory has been specified, an erase cycle will be performed
>              To disable this feature, specify the -D option.
> avrdude.exe: erasing chip
> avrdude.exe: reading input file "H:\Scorpion\Project\3Dprinting\T3D\Mpad\QMK\qmk_firmware\.build\handwired_void9_via_test.hex"
> avrdude.exe: writing flash (22428 bytes):
> 
> Writing | ################################################## | 100% 1.76s
> 
> avrdude.exe: 22428 bytes of flash written
> avrdude.exe: verifying flash memory against H:\Scorpion\Project\3Dprinting\T3D\Mpad\QMK\qmk_firmware\.build\handwired_void9_via_test.hex:
> avrdude.exe: load data flash data from input file H:\Scorpion\Project\3Dprinting\T3D\Mpad\QMK\qmk_firmware\.build\handwired_void9_via_test.hex:
> avrdude.exe: input file H:\Scorpion\Project\3Dprinting\T3D\Mpad\QMK\qmk_firmware\.build\handwired_void9_via_test.hex contains 22428 bytes
> avrdude.exe: reading on-chip flash data:
> 
> Reading | ################################################## | 100% 0.19s
> 
> avrdude.exe: verifying ...
> avrdude.exe: 22428 bytes of flash verified
> 
> avrdude.exe: safemode: Fuses OK (E:CB, H:D8, L:FF)
> 
> avrdude.exe done.  Thank you.
> 
Flash complete
HID console connected: Victor Lucachi VOID9 (F055:3534:0001)
Caterina device disconnected (usbser): Arduino LLC (www.arduino.cc) Arduino Leonardo bootloader (COM18) (2341:0036:0001) [COM18]
HID console disconnected: Victor Lucachi VOID9 (F055:3534:0001)
----------------------

but when I connect it to via it doesn't detected.  Did I make mistake ?

victorlucachi, 2022-05-16:
the macropad does not have official VIA support; you have to download [this file](https://github.com/victorlucachi/void9/blob/master/void9_via_definitions.json) and manually load it into the Configurator.

Enable the Design tab first, from VIA Settings:
[image]

Then load the JSON definitions for the macropad:
[image]

r4ph74, 2022-05-16:
I already upload  the design json file .. still doestn't detected. 

[image]

[image]

victorlucachi, 2022-05-16:
Alright, give me a couple of minutes, I'll do a clean build and test it on my own macropads.

r4ph74, 2022-05-16:
It seems no HID consoles ..detected , when plugged in.

I check on qmk toolbox status at the bottom.

victorlucachi, 2022-05-16:
> 

I can confirm that it wasnt being detected for me either. I made some changes and now the macropad is detected by VIA.

Update your copy of the repository and build the fw again, it should work after.

[image]

r4ph74, 2022-05-16:
ok Thanks .. I will try to update it . I will inform you again .

r4ph74, 2022-05-16:
now I have problem  with qmk toolbox, the com port is not detected. I have reset it several times.. until bootloader com port is out. but the qmk toolbox doesn't detect it . 

the auto flash doesn't work either .

victorlucachi, 2022-05-16:
> 

plug in the macropad, compile and flash the firmware through qmk msys by using the following command: `qmk flash -kb handwired/void9 -km via_test`. it will prompt you to reset the controller; when that happens bridge the reset and ground pins only once.

r4ph74, 2022-05-16:
Yayyy.. finally it works.. great! 

[image]

Fyi  I haven't  built fully yet .. the macropad, it's just the case , and upload the board. 

Now  it's time for soldering ... the switch ..and diode.. 

Thanks alot for  your support Victor.  I will show you when it's done.

victorlucachi, 2022-05-16:
Awesome! have fun with the rest of the build and sorry for the headache.

darknessblade, 2022-07-21:
Currently having the same issue. no matter what I try I cannot get the configurator [QMK MSYS] to compile the Hexfile.

Would you be able to share your hex file, so I can flash it to My void9

darknessblade, 2022-10-22:
The QMK folder of your fork no longer contains the VIA files.

victorlucachi, 2022-10-22:
> 

hey. I did push some changes to my QMK fork, but not sure I follow, what VIA files are you referring to?

darknessblade, 2022-10-22:
I mean the VIA_test Keymap. its no longer in the keymaps folder

https://github.com/victorlucachi/qmk_firmware/tree/dev_void/keyboards/handwired/void9/keymaps

victorlucachi, 2022-10-22:
not an issue, default keymap is via enabled, I'm trying to remove clutter and keep consistency across all macropads and keyboards files.

darknessblade, 2022-10-25:
OK.
____________
With some trying I was able to get a "working" vial  "fork" for the "VOID9"

https://github.com/darknessblade/QMK_VIAL_VOID9

Which makes remapping a whole lot easier, than having to "side"-load it into VIA/VIAL every time with a restart of your PC.
I have nearly all features enabled myself.

victorlucachi, 2022-10-26:
> OK.
> 
> With some trying I was able to get a "working" vial "fork" for the "VOID9"
> 
> https://github.com/darknessblade/QMK_VIAL_VOID9
> 
> Which makes remapping a whole lot easier, than having to "side"-load it into VIA/VIAL every time with a restart of your PC. I have nearly all features enabled myself.

Awesome! I have a fork of Vial too, I dont mention it much, mostly because I added support for my projects sometime last year but I somehow forgot about it. But I agree, it does make things a lot easier than VIA.

[image]

Your reply prompted me to take a look at it again, and after merging the recent changes the VOID9 compiles and flashes without a hitch, the only issue being the firmware size; I have to disable a number of features in order for the fw to fit the good ol Pro Micro.

I have a question though: does Vial lag for a couple of seconds for you too once you start it up/connect the macropad? I dont remember it being like that in the past, not sure if it's something on my end or just a common thing.

darknessblade, 2022-10-26:
Nice, I love the look of the encoder support of VIAL.

What I wonder is if you could use multiple encoders to control volume of independent apps
either trough the volume mixer or eartrumpet
https://eartrumpet.app/

Would make for a fun encoder-pad with 6-9 encoders.

This is one of the things I plan on making down the line

MrReibot, 2023-02-10:
Hello, I have tried to compile, to have a firmware compatible with via, and I have not been successful. Could someone share their .hex, so I can complete my project?

victorlucachi, 2023-02-10:
> Hello, I have tried to compile, to have a firmware compatible with via, and I have not been successful. Could someone share their .hex, so I can complete my project?

hey. did the compiler error out? could you post the output?

darknessblade, 2023-02-10:
> 

As I am using VIAL, the issue is long since resolved. I also prefer VIAL over VIA

victorlucachi, 2023-02-10:
> > > Hello, I have tried to compile, to have a firmware compatible with via, and I have not been successful. Could someone share their .hex, so I can complete my project?
> > 
> > 
> > hey. did the compiler error out? could you post the output?
> > edit: [here's](https://github.com/victorlucachi/void9/files/10704403/handwired_void9_vial.zip) the vial .hex file. you can configure the macropad [here](https://vial.rocks/), make sure to use a compatible browser.
> 
> I'll try it with the .hex you share. I share mistakes, I hope you can help me, thanks.

the .hex file that I linked in my previous response should work both with the via app (you will still have to load the json definitions file) and with vial (plug and play, no definitions required).

did you edit the keymap file at all before attempting to compile? both forks for the VOID9 compile successfully on my end.

MrReibot, 2023-02-10:
> > > > Hola, he intentado compilar, tener un firmware compatible con via, y no he tenido exito. ¿Alguien podría compartir su .hex, para que pueda completar mi proyecto?
> > > 
> > > 
> > > ey. salió el error del compilador? ¿podrías publicar la salida? 
> > > editar: [aquí está](https://github.com/victorlucachi/void9/files/10704403/handwired_void9_vial.zip) el archivo vial .hex. puede configurar el macropad [aquí](https://vial.rocks/) , asegúrese de usar un navegador compatible.
> > 
> > 
> > Lo intentaré con el .hex que compartes. Comparto errores, espero me puedan ayudar, gracias.
> 
> el archivo .hex que vinculé en mi respuesta anterior debería funcionar tanto con la aplicación via (todavía tendrá que cargar el archivo de definiciones json) como con vial (plug and play, no se requieren definiciones).
> 
> ¿Editó el archivo del mapa de teclas antes de intentar compilar? ambas bifurcaciones para VOID9 se compilan con éxito en mi extremo.

Thanks, if it works, I just have a detail on the order of buttons. Could you share me the files to compile vial?

MrReibot, 2023-02-10:
> the .hex file that I linked in my previous response should work both with the via app (you will still have to load the json definitions file) and with vial (plug and play, no definitions required). [...]

Thanks, if it works, I just have a detail on the order of buttons. Could you share me the files to compile vial?

r4ph74, 2023-02-12:
HI Victor, I just reflash the vial one with qmk toolbox. It says complete.  but when I tried reconnect the Vial GUI it doesn't detected.  it seems still have the previous "old" VIA firmware (still have to sideload json file)  

`USB device disconnected (HidUsb): (Standard system devices) USB Input Device (F055:3534:0001)
USB device disconnected (usbccgp): (Standard USB Host Controller) USB Composite Device (F055:3534:0001)
USB device disconnected (HidUsb): (Standard system devices) USB Input Device (F055:3534:0001)
USB device disconnected (HidUsb): (Standard system devices) USB Input Device (F055:3534:0001)
USB device disconnected (HidUsb): (Standard system devices) USB Input Device (F055:3534:0001)
Caterina device connected (usbser): Arduino LLC (www.arduino.cc) Arduino Leonardo bootloader (COM7) (2341:0036:0001) []
Attempting to flash, please don't remove device
Flash complete
Caterina device disconnected (usbser): Arduino LLC (www.arduino.cc) Arduino Leonardo bootloader (COM7) (2341:0036:0001) []
USB device connected (HidUsb): (Standard system devices) USB Input Device (F055:3534:0001)
USB device connected (usbccgp): (Standard USB Host Controller) USB Composite Device (F055:3534:0001)
USB device connected (HidUsb): (Standard system devices) USB Input Device (F055:3534:0001)
USB device connected (HidUsb): (Standard system devices) USB Input Device (F055:3534:0001)
USB device connected (HidUsb): (Standard system devices) USB Input Device (F055:3534:0001)
USB device disconnected (HidUsb): (Standard system devices) USB Input Device (F055:3534:0001)
USB device disconnected (usbccgp): (Standard USB Host Controller) USB Composite Device (F055:3534:0001)
USB device disconnected (HidUsb): (Standard system devices) USB Input Device (F055:3534:0001)
`
What should I do ?

victorlucachi, 2023-02-12:
>  What should I do ?

hello. how did you conclude that the controller is using the old firmware?

are you using the vial app or the web client? if it's the latter, make sure that you are using a compatible browser that supports the webhid api, and that you're granting access for the macropad.

[image]

I test flashed a pro micro and it's all good as far as I can tell.

[image]
[image]

r4ph74, 2023-02-12:
hi Victor 

I just reflash the second times. and it works now.. but the colum is mixed with the middle one.. have to rethinkering to modif the layout or the pinout in coding... .. thanks for the response.
[image]

victorlucachi, 2023-02-12:
> hi Victor
> 
> I just reflash the second times. and it works now.. but the colum is mixed with the middle one.. have to rethinkering to modif the layout or the pinout in coding... .. thanks for the response.

yes. I apologize for that, I made a couple of changes in the firmware and the wiring matrix, which are mentioned in the readme. you are using the old pinout, that's why the columns are not in the correct order, give me a couple of minutes and I'll reply with a different .hex file

r4ph74, 2023-02-12:
Perfect ! .. Thanks Victor .. you're  the best.
[image]

victorlucachi, 2023-02-12:
awesome, glad it works!

MrReibot, 2023-02-12:
> > hi Victor
> > I just reflash the second times. and it works now.. but the colum is mixed with the middle one.. have to rethinkering to modif the layout or the pinout in coding... .. thanks for the response.
> 
> yes. I apologize for that, I made a couple of changes in the firmware and the wiring matrix, which are mentioned in the readme. you are using the old pinout, that's why the columns are not in the correct order, give me a couple of minutes and I'll reply with a different .hex file
> 
> edit: I've attached the .hex for the old pinout here: [handwired_void9_vial_old_pinout.zip](https://github.com/victorlucachi/void9/files/10715663/handwired_void9_vial_old_pinout.zip) please let me know if it works.

Thank you, you are the best, I had the same problem.

[image]