victorlucachi/void9 #8: EC11 encoder pinout diagram

r4ph74, 2023-02-12:
HI Victor  It's me again . 

I want to make a new one with encoder since I have successfully run the vial.  But I still confuse with the pinout. 

the encoder have 5 pin  2 + 3 . I know the 2 pin one is for colom and other for the row to act, both pin act as  a button .  The other 3 pin is for rotary up and down.  the 2 pin is already known as you wrote, but how about the last pin ? where does it go ?  does it go to the same colom as the other 2 pin ? 

if you have the diagram that would be delightful. 

Thanks in Advance 

Rafael

victorlucachi, 2023-02-12:
hey Rafael,

I dont have any diagrams, but you got it right.

the side with two pins gets wired into the key matrix. on the other side, out of the three pins the upper and lower ones are connected according to the pinout in the readme (pin A to B2 and pin B to B3), while the third middle pin gets connected to ground on the mcu.

r4ph74, 2023-02-12:
Ahh.. Thanks Victor!..

r4ph74, 2023-02-17:
hi Victor  it's me again . 

 I have soldered Pin B1 (15)  gnd (ground) and B2 (16) .. but seems the vial doesn't recognise it ... 

the press button works good as usual .  

did I miss something ?

r4ph74, 2023-02-17:
Thanks VIctor , it's working  now. I missed how to read the pinout  you mentioned  in readme.

victorlucachi, 2023-02-17:
> Thanks VIctor , it's working now. I missed how to read the pinout you mentioned in readme.

no worries, glad it's working!

the pinout table is a bit confusing, I'll try to make things a little clearer in the future, and add mcu/encoder diagrams and highlight the used pins.

r4ph74, 2023-02-17:
That would be great. Thanks Victor.

closed 2023-06-13 by victorlucachi