victorlucachi/void9 #9: Waveshare rp2040 controller doesn't fit

Sparticuz, 2023-06-23:
The chips on the bottom of the waveshare rp2040 won't allow it to slide into the controller slides. Seems like a 1 or 2 mm spacer needs to be cut out of the middle

[image]
[image]
[image]

victorlucachi, 2023-06-23:
this thing will have close to no structural integrity at all 😅. please let me know if it works.

[waveshare-rp2040.zip](https://github.com/victorlucachi/void9/files/11849325/waveshare-rp2040.zip)

Sparticuz, 2023-06-23:
Lol, will do. It looks like about half that could probably be put back. Though, also once it's screwed in, that will add back a ton of rigidity. I really wish i had my calipurs here at work. :/

Sparticuz, 2023-06-28:
It worked much better, the only improvement would be to make the screw holes smaller because the m2 insert i was using was quite a bit smaller than the opening. I had to wiggle the insert around inside the hole to get it to melt the sides.

victorlucachi, 2023-06-28:
sweet. the inserts that I'm using (also linked in the bom) have an outer diameter of 3.5mm; they should fit the part.

Sparticuz, 2023-06-28:
I used the same inserts on the top part and they fit perfect, but on the controller mount the hole was much larger. Everything is working well though! thank you for altering the stl!

closed 2023-06-28 by Sparticuz