victorlucachi/void9 #11: Contact

Mboehmlaender, 2024-03-03:
Hi there,
 
unfortunately, i could Not contact you via Email, so i just opened an issue. You can delete it afterwards, if you wish. 

Some time ago, i stumbled upon your VOID9 macropad and i fell in love with it. I love the tiny, classy and simple design so i made a first print and wired it and it worked like a charme. 
 
Then i decided to take it a step further and designed the VOID9.1. I forked your repo and modified the prints and added RGB-Support and rewrote some parts of the code. Now it's all done and i completed the readme on github and published the repo. I wanted to ask, if you could have a look at at https://github.com/Mboehmlaender/kb-void9.1 and tell me, if there are any licence issues in the code or the repo itself, so i can adjust it to give you the credit, you deserve. And maybe a star would be nice. 
 
Thanks in advance und greetings from bavaria.
 
Michael

victorlucachi, 2024-03-03:
Hey Michael, and thank you for the kind words.

Everything seems in order, apart from that one small typo (my last name, in the [_About the Project_](https://github.com/Mboehmlaender/kb-void9.1?tab=readme-ov-file#about-the-project) section of the readme), and the license (my models are released under the [CC BY-NC-SA](https://creativecommons.org/licenses/by-nc-sa/4.0/) license. you should go over it, but to keep it short you would have to release your mods under the same license).

Cool stuff, nicely documented, and I appreciate you contacting me. Thanks again!

Mboehmlaender, 2024-03-04:
Hi Victor,

thanks for starring my fork, i really appreciate it. I made the modifications you stated and i hope that GH updates the license in the frontend soon.

I chose GPL license because there were GPL notes in the firmware (config.h) in your VIAL fork. Does this relate to the VIAL/QMK itself? Then i need to change it back as well.

Greetings